larus-breeze/hw_sensor #5: RJ45 Positionen

MaxBaex, 2023-02-14:
Die RJ45 Buchsen stehen zu nahe am Rand für das schmale Gehäuse 1455J1201.  Es ist mit diesem Gehäuse auch kein Platz für SMA Buchsen auf der Rückseite.

MaxBaex, 2023-02-18:
Steckeralternativen evaluieren:  ix Industrial     kleiner RJ45 Nachfolger.  Mouser: https://www.mouser.de/ProductDetail/HARTING/09452812560?qs=lc2O%252BfHJPVb7MWmafU5g%252Bw%3D%3D

floriangaller, 2023-02-20:
> Steckeralternativen evaluieren:  ix Industrial     kleiner RJ45 Nachfolger.  Mouser: https://www.mouser.de/ProductDetail/HARTING/09452812560?qs=lc2O%252BfHJPVb7MWmafU5g%252Bw%3D%3D  
> 
> Alternativ auf einen RJ45 Anschluss reduzieren und RS232 /  CAN beides auf diesem zur Verfügung stellen.  Gibt es Use-Cases für mehr als einen RS232 Anschluss?

Das würde ich nicht machen... RJ45 ist standard für Serial NMEA Verbindungen in Segelfliegern, leicht erhältlich, veriegelt,... 

Was man IMHO überlegen kann...
 Es gibt low proflie RJ45 die könnten das Höhenproblem lösen
 Den CAN könnte man auf einen anderen Stecker legen.

Der Nachteil an den RJ45 ist eigentlich,dass sie halt nicht sauber spezifiziert sind... Vorallem im Bezug auf Kabelbrand... man sollte halt in einem Cockpit nix verbauen was von selbst weiterbrennt und eigentlich auch nix was übermäßig Rauch eintwickelt... (GFK/CFK ist da vermutlich ausgenommenn :-))

Wenn man in Richtung Luftfahrt konformer Stecker gehen will, dann kann man sich die MIL Stecker von Amphenol ansehen... Aber die sind auch relativ groß und teuer... aber machen beim Anstecken einen sehr wertigen eindruck.

Aber ich denke nicht das uns der standard Segelflieger dankbar ist wenn wir für uns den Standard ändern...

floriangaller, 2023-02-20:
> Alternativ auf einen RJ45 Anschluss reduzieren und RS232 /  CAN beides auf diesem zur Verfügung stellen.  Gibt es Use-Cases für mehr als einen RS232 Anschluss?

Ich denke schon, dass es dafür use cases gibt:

Eigentlich hat man ja fast nie genug Serielle im Flieger...
XPDR will ein GPS mit komischer Baudrate manchmal auch komischen Format
ELT will GPS mit komischer Baudrate
Funk kann man fernsteuern
IGC Höhe aus dem Logger, Task programmieren, igc file auslesen...
Vergleich mit anderem Vario und daher Aufzeichnung des NEMA Streams...
Anschluss xc vario und openvario

Bei CAN bin ich leider eher blank aber da kann ich mir auch eine Vielzahl an Sensoren vorstellen:
Wölbklappen, Abgriff der Steuerung, Temperatur, Sonneneinstrahlung, TEK Druck, Fahrwerk, Knüppel, Bremsklappe, ...

Ich sehe das Ganze schon eher als Plattform auf der dann weitere Entwicklungen aufgesetzt wereen können, und da hat man halt meist zu wenige Interfaces... Vlt brauchts da zwei Versionen, wobei das macht halt leider auch alles wieder sehr aufwändig... 
Oder Steckerbelegung über Bestückungsänderumg... aber das ist halt wieder super gefährlich mit unbedarften Benutzern

MaxBaex, 2023-02-20:
Ich denke auch, das wir keinen neuen Standard für die "seriellen" Anschlüsse erfinden sollten. Hier ist RJ45 einfach ein bewährter Standard. Denkbar wäre anstelle des CAN-RJ45 ein kleinerer 4 poliger Stecker, für Spannung und CAN. Der CAN-RJ45 kann aktuell in einen seriellen RJ45 umgewandelt werden. Das Feature sollte denke ich bestehen bleiben. 

Vielleicht passen 2 SMA Anschlüsse mit eng gepackten RJ45 Low-Profilen (z.B. RJE051880310) und einem evtl. etwas geschrumpften Druckanschluss noch in das Gehäuse.

floriangaller, 2023-02-20:
> Ich denke auch, das wir keinen neuen Standard für die "seriellen" Anschlüsse erfinden sollten. Hier ist RJ45 einfach ein bewährter Standard. Denkbar wäre anstelle des CAN-RJ45 ein kleinerer 4 [...]

Viel schmälere Stecker zu finden wird vermutlich schwierig.

Ix wäre okay, sonst könnte man auch noch einen M8 nehmen. Z.B. Amphenol M8AS-04PMMR-SF8001. Aber ob man da wirklich platz spart muss man sich genau ansehen, denn die stecker haben ja auch einen mindestabstand damit man mit den Kabeln kein Problem bekommt... da hilft es ungemein das ganze voher in ein Cad zu werfen. Ich befürchte viel kleiner als rj45 sind dann beide nicht.

MaxBaex, 2025-01-01:
Wir werden bis auf weiteres bei RJ45 bleiben, daher schließe ich dieses Ticket.

closed 2025-01-01 by MaxBaex