larus-breeze/hw_sensor #18: MTi-1 HW V5

henrik-the, 2024-06-26:
Hi,

did anybody already test the HW V5 of the MTi-1 Module? Is it compatible with the current software?
https://www.mouser.de/ProductDetail/Movella-Xsens/MTi-1-5A-T?qs=3Rah4i%252BhyCG%252BD2yTE%2FxmKQ%3D%3D&mgh=1&vip=1&utm_id=20979042634&gad_source=1&gclid=CjwKCAjwgdayBhBQEiwAXhMxtlsLLCMI9tfVR8Dzdq2Rfec0irS5VLUbWIleFCXaXezehbrgSJYrmxoCJrcQAvD_BwE

The Migration Guide tells, that it should be compatible:
https://base.movella.com/s/article/Migration-to-Hardware-Version-5-of-the-MTi-1-series?language=en_US

Laca81, 2024-11-02:
I just built one with MTi-1-5A-T and it works fine. This is my first Larus build so I don't know how it works with the previous version of this IMU.

realtimepeople, 2024-11-03:
Hi guys, great to hear that your have managed to build a Larus on your own.
I would be very much interested in some *.f37 log files from your sensors to check if the new IMU version is more precise than the old ones. Can you please contact me via the official larus mail larus@e.mail.de ?

henrik-the, 2024-11-28:
Hi,
I have build my Larus with the "old" IMU. So i think I can't help.

closed 2025-02-23 by MaxBaex

invrtd, 2025-03-20:
@realtimepeople Did you get any of the wanted files? If not I would not mind sending some after I get mine built with the newer IMU.

realtimepeople, 2025-03-21:
No, up to now I did not get any log files with the new IMU.

Laca81, 2025-03-21:
I am totally sure that I sent you a log. Should I resend it or a new one?

realtimepeople, 2025-03-21:
Oh, sorry. 
Yes, I have seen one of your short logs some time ago.

It would be interesting to see several minutes of a log file,
measured on ground, while circling and flying straight.