larus-breeze/hw_sensor #27: Release 2.2 BOM and CPL

invrtd, 2025-02-25:
I've been planning on building my own Larus for a bit now and the new release has gotten me moving on the project again and I've noticed a few issues.

Problems I can fix myself but need to be fixed in the release,
- The usb-c headers J4 and J5 have incorrect translation
- - The positions for RJ-45 j3 and j8 have incorrect rotation and translation.
- BOM says MS5611-01BA for U7 while kicad has MS5611-01BA03 as the Mfr. No
- This is a nit the capacitors for C37 and C60 are out of stock and substituted with  LCSC# C913910
-
And I could find parts that are close but I want to make sure my substitutes should be okay 
- The blue LEDs are out of stock with no equivalent so LCSC# C125103 seems close enough with lower wattage 
- The HSCMRRV001PG2A3 is EOL and seems to be a 1:1 replacement HSCMRRV001PD2A3 ???

MaxBaex, 2025-02-25:
- any blue LED in a 0603 package should do.  You might want to adjust the series resistor depending on the desired current.  
- HSCMRRV001P**D**2A3 won't work as it is a +- differential, instead of a gauge version. Choosing this one would half the resolution and requires firmware changes. We do not recommend this. 
- If changing capacitors make sure you chose the same voltage rating, value and type(e.g. 16V 100nF **X7R**)
- I might fix incorrect translation or rotations in a future release   
- MS561101BA03-50 is the correct one

invrtd, 2025-02-25:
Do you have any recommendations on sourcing the HSCMRRV001PG2A3? I am having a hard time finding a source to purchase in small quantities.

MaxBaex, 2025-02-28:
Mouser states HSCMRRV001PG2A3  as EOL.  Farnel and RS-online does not. Currently we do not see the need for changing this component.

closed 2025-02-28 by MaxBaex